robotpy/robotpy-wpilib #457: DoubleSolenoids Break Simulation

adam-mcdaniel, 2018-02-24:
When I create a DoubleSolenoid object like this: wpilib.DoubleSolenoid(10, 0, 1), I get a KeyError: 10 error. When I take the 10 out and use: wpilib.DoubleSolenoid(0, 1), I get this error: 

```

Traceback (most recent call last):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\robotbase.py", line 168, in main
    robot.startCompetition()
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\iterativerobot.py", line 59, in startCompetition
    self.loopFunc()
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\iterativerobotbase.py", line 181, in loopFunc
    LiveWindow.updateValues()
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\livewindow.py", line 333, in updateValues
    component.builder.updateTable()
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\sendablebuilder.py", line 69, in updateTable
    prop.update(prop.entry)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\sendablebuilder.py", line 188, in <lambda>
    updater = None if getter is None else lambda entry: entry.setString(getter())
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\doublesolenoid.py", line 178, in <lambda>
    builder.addStringProperty("Value", lambda: self.get().name[1:], self._valueChanged)
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\wpilib\doublesolenoid.py", line 145, in get
    if hal.getSolenoid(self.forwardHandle):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\hal\functions.py", line 58, in outer
    rv = _inner(*args, status=status, **kwargs)
  File "<string>", line 5, in getSolenoid
AssertionError: Internal Error: Invalid return value from getSolenoid (check was: isinstance(return_value, bool)); value=None, type=NoneType

Locals at innermost frame:

{ 'return_value': None,
  'solenoidPortHandle': <SolenoidHandle at 0x62db0d0 mod=0 pin=0>,
  'status': c_long(0)}
```

Am I doing something wrong or is it broken?

auscompgeek, 2018-02-25:
I actually can't figure out why this would be happening. Indeed, getting the value of an unset solenoid is even in the tests already. My team has also been using pneumatics fine this year, albeit only using the default module 0.

You don't happen to have a physics.py which is doing weird things it shouldn't be, do you?

virtuald, 2018-02-25:
I was just trying to reproduce this, I was not able to reproduce it locally. Are you sure you have the latest RobotPy libraries installed?

virtuald, 2018-02-25:
Also, can you provide a simple example that reproduces this behavior? Thanks!

adam-mcdaniel, 2018-02-25:
I found the problem. I have a library called "subsystems.py" it creates instances of all the objects needed for the IterativeRobot. I imported it before I defined my class, so I kept getting errors. If I create functions that return my Chassis, Launcher, and Elevator objects and use those to create my objects in the robot_init, the errors go away.

closed 2018-02-25 by virtuald