robotpy/robotpy-wpilib #462: RuntimeError: dictionary changed size during iteration

ariovistus, 2018-03-07:
is being thrown in the unittests of our command based robot code

example build:
https://travis-ci.org/Retro3223/robot2018/builds/347137201?utm_source=github_status&utm_medium=notification

auscompgeek, 2018-03-07:
This is very odd… I can't find any writes to LiveWindow.components that aren't protected by the mutex.

virtuald, 2018-03-08:
Well, you guys are doing something weird, during tests there are ~1500 items in LiveWindow

virtuald, 2018-03-08:
Looks like everytime you call getJoystick, a new Joystick and JoystickButton are created... and for whatever reason JoystickButton is Sendable. Doesn't explain the error, but you should probably fix that.

virtuald, 2018-03-08:
If the user does something silly like create a component in the middle of getting values for LiveWindow, then this error occurs. Fixed it so we just make a copy of the dictionary values.

You should probably not create so many JoystickButton objects.

commit 0b0285fe (2018-03-08): LiveWindow: grab copy of components dict to prevent unintentional errors

- User code could potentially do something like create a new component in their updateTable
- Fixes #462
closed 2018-03-08 by virtuald