robotpy/robotpy-wpilib #474: Update RobotPy for 2019

auscompgeek, 2018-08-19:
Opening this meta-issue early this year for the PyCon AU sprints (next week).

For those of you new to this project, every year we have to update RobotPy's source code to match whatever changes they made to WPILib. It's a bit time consuming and a very manual process.

Last year was a bit of an outlier due to the large changes in WPILib, but in the past we mostly ported things over by the end of October.

We have some tools that help out with this. If you're interested in helping out then check out our porting guide at <http://robotpy.readthedocs.io/en/latest/dev/porting.html>. If the instructions don't make sense, please comment here so we can update them.

Note that the summer hasn't ended in the US yet, and the WPILib team do still have a couple of outstanding projects, so things are still subject to significant change, but I think the bulk of the work has died down now.

virtuald, 2018-08-20:
One big thing that needs to be done across our repos is update git-source-track lines to match the new repo, as they now have a mega repo which will mess with our tracking stuff...

virtuald, 2018-10-24:
Sorry... I've been super busy at work this year.

auscompgeek, 2018-10-25:
Hey, no biggie. We all have lives, and no one person should have to take on all the work of a semi-large project.

I don't think we've actually asked for help on the usual channels yet, so that'd probably be good to do.

auscompgeek, 2018-12-21:
There are also various tasks listed on https://github.com/orgs/robotpy/projects/3. I've tried to order them based on rough priority.

auscompgeek, 2018-12-25:
Dibs on Watchdog.

auscompgeek, 2018-12-30:
Before anyone decides to jump on the MotorSafety changes, it looks like upstream have decided to revert MotorSafety using Watchdog: wpilibsuite/allwpilib#1517

auscompgeek, 2019-01-02:
Incoming changes to MotorSafety, LiveWindow, IterativeRobotBase, RobotState, and command.PIDSubsystem.

auscompgeek, 2019-02-22:
Other than Ultrasonic, I think this is pretty much done.

closed 2019-02-22 by auscompgeek