robotpy/robotpy-wpilib #537: Port CAN API

auscompgeek, 2018-12-18:
There have been mentions that the SPARK MAX CAN library will be using the WPILib CAN classes, so this should probably happen soon-ish.

robobenklein, 2019-01-06:
Ri3D Team here with 6 Spark MAXs and we haven't been able to get the MAXs to show up in CAN with any of the CTRE tools.

auscompgeek, 2019-01-06:
@robobenklein That's probably because CTRE's tools don't support the Spark MAX? Not that we're affiliated with CTRE nor REV.

virtuald, 2019-01-07:
Looks like the Spark MAX software isn't available yet: https://www.chiefdelphi.com/t/rev-spark-max-neo-apis-info/337013

Brandutchmen, 2019-02-01:
^^ Our team is looking to use SPARK Max ESCs. If we can CAN them, then that would be wonderful.
Otherwise, we could just run them as a PWM connection. We just would be sacrificing the encoder support.

Thank you guys so much!

auscompgeek, 2019-02-02:
@Brandutchmen See:
- https://www.chiefdelphi.com/t/robotpy-spark-max-bindings/339180
- https://robotpy.readthedocs.io/en/stable/install/rev.html
- https://robotpy.readthedocs.io/projects/rev/en/latest/api.html

virtuald, 2020-01-19:
I believe this no longer applies now that we're using wpilibc.

closed 2020-01-19 by virtuald