robotpy/robotpy-wpilib #538: Skip tests for deprecated classes

twinters007, 2018-12-18:
No reason to test Robotdrive since we aren't updating it

virtuald, 2018-12-18:
Yeah, but the code still exists, so it needs to have tests to ensure we don't accidentally break it. This seems like a better fit:  https://docs.pytest.org/en/latest/warnings.html

closed 2018-12-18 by virtuald

twinters007, 2018-12-18:
But we aren't gonna change the code, so can't just skip them in tests? It'll speed things up.

virtuald, 2018-12-18:
We change things that the code depends on.

twinters007, 2018-12-18:
Doesn't being deprecated mean that we don't guarantee those things will work with new changes?

virtuald, 2018-12-18:
No, it means that it will eventually be removed so you shouldn't use it.

One thing we should definitely do is remove deprecated stuff from the examples. But as long as the code is here, it needs to stay tested because someone could use it.