robotpy/robotpy-wpilib #562: Fix tag deploy

virtuald, 2019-01-05:
Seems to only upload some pypi artifacts, not all of them.

virtuald, 2019-01-05:
Additionally, using wheels doesn't bring in our dependencies. This applies to pyfrc also. I've deleted the wheels from pypi for now.

auscompgeek, 2019-01-06:
I built the wpilib wheel locally and it declares dependencies correctly, so I'm not sure what's happening here.

With pyfrc though, the CI configuration is actually broken - the setup.py doesn't declare the dependencies due to the environment.

auscompgeek, 2019-01-06:
With the HAL packages not being uploaded to PyPI, turns out Travis CI counterintuitively only creates single jobs for each entry in `matrix.include`, even if you use a list for values like `env`. Hmm.

I'll have to double-check the Travis CI documentation, but I seem to be under the impression that jobs will inherit the settings of the previous declared job. I might test this evening.

virtuald, 2019-01-06:
When I tried doing a 'stock' install of non-whl pyfrc using an empty virtualenv, the hal-base package was missing:

> Successfully installed atomicwrites-1.2.1 attrs-18.2.0 more-itertools-5.0.0 pint-0.8.1 pluggy-0.8.0 py-1.7.0 pyfrc-2019.0.0 pynetconsole-2.0.2 pynetworktables-2019.0.0 pytest-4.0.2 robotpy-hal-sim-2019.0.0 robotpy-installer-2019.0.0 robotpy-wpilib-utilities-2019.0.0 six-1.12.0 wpilib-2019.0.0\

Once I deleted the whl from pypi, it worked as expected.

auscompgeek, 2019-01-06:
Only the robotpy-hal-sim and robotpy-hal-roborio packages actually declare a dependency on robotpy-hal-base, by the looks of things.

I can't find anything that would suggest that the robotpy-hal-sim package's dependency on robotpy-hal-base would've disappeared.

auscompgeek, 2019-01-06:
I have another idea for deduplicating the deploy. You can make deploys conditional on an arbitrary bash condition. (Additionally there's a `$TRAVIS_BUILD_STAGE_NAME`, but probably won't be necessary.)

PR #565 (2019-01-07): Fix deploy in Travis CI config
Bonus: Make the format-check job run in parallel too.

Closes: #562
closed 2019-01-07 by virtuald