robotpy/robotpy-wpilib #567: Camera Server Crashes

JDuskey, 2019-01-08:
This error occurs when running the quick vision example code:

robot.py:3: DeprecationWarning: IterativeRobot is deprecated, use TimedRobot instead.
  class MyRobot(wpilib.IterativeRobot):
20:22:27:231 INFO    : wpilib              : WPILib version 2019.0.1
20:22:27:232 INFO    : wpilib              : HAL base version 2019.0.1; roboRIO platform version 2019.0.1
20:22:27:233 INFO    : wpilib              : HAL library version 2019.1.1
20:22:27:234 INFO    : wpilib              : robotpy-cscore version 2019.0.0
20:22:27:236 INFO    : faulthandler        : registered SIGUSR2 for PID 1831
20:22:27:238 INFO    : nt                  : NetworkTables initialized in server mode
20:22:27:252 INFO    : wpilib.cs           : Launching CameraServer process
20:22:27:281 ERROR   : wpilib.ds           : Unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/wpilib/robotbase.py", line 186, in main
    robot.startCompetition()
  File "/usr/local/lib/python3.7/site-packages/wpilib/iterativerobot.py", line 65, in startCompetition
    self.robotInit()
  File "robot.py", line 5, in robotInit
    wpilib.CameraServer.launch()
  File "/usr/local/lib/python3.7/site-packages/wpilib/cameraserver.py", line 84, in launch
    args, close_fds=True, stdin=subprocess.PIPE, cwd="/home/lvuser/py"
  File "/usr/local/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/local/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/lvuser/py': '/home/lvuser/py'

Locals at innermost frame:

{ 'args': ['/usr/local/bin/python3', '-m', 'cscore'],
  'c2pread': -1,
  'c2pwrite': -1,
  'child_exception_type': <class 'OSError'>,
  'child_exec_never_called': True,
  'close_fds': True,
  'creationflags': 0,
  'cwd': '/home/lvuser/py',
  'devnull_fd': None,
  'env': None,
  'env_list': None,
  'err_filename': '/home/lvuser/py',
  'err_msg': "No such file or directory: '/home/lvuser/py'",
  'errno_num': 2,
  'errpipe_data': bytearray(b'OSError:2:noexec'),
  'errpipe_read': 22,
  'errpipe_write': 23,
  'errread': -1,
  'errwrite': -1,
  'exception_name': bytearray(b'OSError'),
  'executable': b'/usr/local/bin/python3',
  'executable_list': (b'/usr/local/bin/python3',),
  'fds_to_keep': {23},
  'hex_errno': bytearray(b'2'),
  'low_fds_to_close': [],
  'orig_executable': '/usr/local/bin/python3',
  'p2cread': 20,
  'p2cwrite': 21,
  'part': b'',
...

Error at /usr/local/lib/python3.7/subprocess.py.1522:_execute_child: Unhandled exception

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/wpilib/robotbase.py", line 186, in main
    robot.startCompetition()
  File "/usr/local/lib/python3.7/site-packages/wpilib/iterativerobot.py", line 65, in startCompetition
    self.robotInit()
  File "robot.py", line 5, in robotInit
    wpilib.CameraServer.launch()
  File "/usr/local/lib/python3.7/site-packages/wpilib/cameraserver.py", line 84, in launch
    args, close_fds=True, stdin=subprocess.PIPE, cwd="/home/lvuser/py"
  File "/usr/local/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/local/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
  File "/usr/local/lib/python3.7/site-packages/wpilib/robotbase.py", line 186, in main
    robot.startCompetition()
  File "/usr/local/lib/python3.7/site-packages/wpilib/iterativerobot.py", line 65, in startCompetition
    self.robotInit()
  File "robot.py", line 5, in robotInit
    wpilib.CameraServer.launch()
  File "/usr/local/lib/python3.7/site-packages/wpilib/cameraserver.py", line 84, in launch
    args, close_fds=True, stdin=subprocess.PIPE, cwd="/home/lvuser/py"
  File "/usr/local/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/local/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/lvuser/py': '/home/lvuser/py'

20:22:27:295 WARNING : wpilib.ds           : Robots should not quit, but yours did!
Robots should not quit, but yours did!
20:22:27:296 ERROR   : wpilib.ds           : The startCompetition() method (or methods called by it) should have handled the exception above.
The startCompetition() method (or methods called by it) should have handled the exception above.

JDuskey, 2019-01-08:
This did work, however now I am having an issue getting a link to the stream i need. All i am trying to do is get the mjpg of 1280x720 but it is giving me the smallest possible pixel ratio, and I do not know how to change. Thank you!

auscompgeek, 2019-01-08:
cscore will default to the lowest resolution for a USB camera. If you wish to use a higher resolution, you will need to use a custom script to configure the camera settings.

I would not recommend running a camera at 720p - a lower resolution is usually more useful. Be mindful of the lower bandwidth limit this year too.

virtuald, 2019-01-08:
I believe if you connect to the stream via shuffleboard, it can ask for a different resolution.

auscompgeek, 2019-01-08:
@virtuald Changing the stream resolution is a resize in software - the camera will still be running at the same resolution as configured.

virtuald, 2019-01-08:
Is that true? I thought there was a way for shuffleboard to ask for a reconfiguration if the user didn't explicitly set one.

You can also ask for a different resolution by accessing `/?action=stream&resolution=1280x720` (which might be what shuffleboard does)... but as @auscompgeek says, you really shouldn't do that. Instead, stretch the smaller image out using your dashboard software.

virtuald, 2019-01-09:
Anyways, user's issue fixed.

closed 2019-01-09 by virtuald