robotpy/robotpy-wpilib #574: Shuffleboard adding widget fails

JDuskey, 2019-01-26:
`   display = shuffle.Shuffleboard()
        self.speed = wpilib.BuiltInAccelerometer()
        self.speed.initSendable
        accel = display.getTab("Drive")\
            .add("Accelerometer", self.speed)\
            .withWidget("Built In Accelerometer")\
            .getEntry()`

this returns an error saying add requires two positional arguments but three were given. No matter what I do I can't get any sensors to display in shuffleboard, but I can get limelight values to display. I'd like to be able to use widgets.

auscompgeek, 2019-01-29:
I've ported across the Shuffleboard example: https://github.com/robotpy/examples/blob/master/shuffleboard/robot.py

Hopefully it should be a lot clearer how to use the Shuffleboard class.

closed 2019-01-29 by auscompgeek

JDuskey, 2019-01-30:
Thank you for porting it, sorry for the delayed response, life has been hectic and I've been having a time trying to get everything working. Your example makes it clearer though.