robotpy/robotpy-wpilib #579: I2C Error

nu11us, 2019-01-31:
I am having errors appear when I try to access an Arduino through I2C using the `wpilib.I2C` class. Attached is the error and code for both the RoboRIO and Arduino. Trying the `transaction` and `read` methods return the same `OSError`.

```
 Traceback (most recent call last): 
   File "/usr/local/lib/python3.7/site-packages/wpilib/robotbase.py", line 186, in main 
     robot.startCompetition() 
   File "/usr/local/lib/python3.7/site-packages/wpilib/samplerobot.py", line 177, in startCompetition 
     self.operatorControl() 
   File "/usr/local/lib/python3.7/site-packages/magicbot/magicrobot.py", line 348, in operatorControl 
     self.onException(forceReport=True) 
   File "/usr/local/lib/python3.7/site-packages/magicbot/magicrobot.py", line 346, in operatorControl 
     self.teleopInit() 
   File "/home/lvuser/py/robot.py", line 48, in teleopInit 
     data = i2c.readOnly(16) 
   File "/usr/local/lib/python3.7/site-packages/wpilib/i2c.py", line 193, in readOnly 
     return hal.readI2C(self.port, self.deviceAddress, count) 
   File "/usr/local/lib/python3.7/site-packages/hal/functions.py", line 543, in readI2C 
     raise IOError(errno, _os.strerror(errno)) 
 OSError: [Errno 0] Success 
  
  
 Locals at innermost frame: 
  
 { 'buffer': <hal.functions.c_ubyte_Array_16 object at 0xb442b6c0>, 
   'count': 16, 
   'deviceAddress': 9, 
   'errno': 0, 
   'port': 0, 
   'rv': -1} 
```

**Relevant Robot Code**
```
import wpilib
def teleopInit(self):
    i2c = wpilib.I2C(0, 9)
    data = i2c.readOnly(16)
    print(data)
```
**Relevant Arduino Code**
```

#include <Wire.h>

byte array1[16];
void setup() {
  Wire.begin(9);
  Wire.onReceive(requestEvent);
}

void requestEvent() {
  Wire.write(array1,16);
} 
void loop(){
  for (int i=0; i<16; i++){
    array1[i] = (byte)(1);
  }
  delay(100);
}
```

auscompgeek, 2019-02-01:
>  i2c = wpilib.I2C(0, 9)

What on earth is 0?

auscompgeek, 2019-02-01:
From the [I2C.readOnly() docs](https://robotpy.readthedocs.io/projects/wpilib/en/latest/wpilib/I2C.html#wpilib.i2c.I2C.readOnly):

> This method does not write any data to prompt the device.

Since your Arduino is waiting for data before it writes data to the I2C bus, it never writes anything.

This seems to otherwise be a duplicate of #438.

nu11us, 2019-02-01:
> > i2c = wpilib.I2C(0, 9)
> 
> What on earth is 0?

Port: kOnboard (wpilib.I2C.Port is an enum with kOnboard=0)

auscompgeek, 2019-02-01:
Yeah, passing a plain integer to functions that are documented to take enums is deprecated. Very deprecated.

Use the actual enums.

nu11us, 2019-02-01:
Yeah, my bad on that.