robotpy/robotpy-wpilib #589: LiveWindow taking way too much time

virtuald, 2019-02-20:
This is a continuation of #441, where LiveWindow is taking a reported 10-20ms+ to do it's thing. Needs some investigation (again).

Recommend that teams disable it for now until we can get to the bottom of this.

auscompgeek, 2019-02-21:
My team instantiates a bunch of DigitalInputs, Solenoids, and DoubleSolenoids. Dustin has a hunch that we might be looking at ctypes overhead here.

virtuald, 2019-02-25:
A magicbot program with 10 digital inputs and 8 PWM Talons takes 5ms for LiveWindow to update, so looks like I have something to go on now.

virtuald, 2019-02-25:
Here's a copy of the cachegrind file from that run. 80 seconds in teleoperated mode, magicbot, not really doing anything. Using latest tags on pypi:

```
pip                      18.1                
pprofile                 2.0.2               
pynetworktables          2019.0.0            
robotpy-hal-base         2019.1.2            
robotpy-hal-roborio      2019.1.2           
robotpy-wpilib-utilities 2019.0.6            
setuptools               40.6.2              
wheel                    0.33.1              
wpilib                   2019.1.2  
```
[cachegrind.out.gz](https://github.com/robotpy/robotpy-wpilib/files/2899133/cachegrind.out.gz)

virtuald, 2019-02-25:
Looking at the profile data, it looks like most of the gain can come from improvements in ntcore. Will play with that this week.

virtuald, 2019-02-28:
I've been chasing this for a few days now, I think I've caught all the low hanging fruit. It's definitely better (need to make some releases in the morning), but it's still not great yet.

It's clear from this that a pure python pynetworktables is just very slow. There seems to be a lot of lock contention, and it takes awhile to set a value over and over again. For an example of lock contention, deleting the fmsinfo updates in the driverstation seems to decrease my LiveWindow period by about 1ms.

virtuald, 2020-01-19:
OBE.

closed 2020-01-19 by virtuald