robotpy/robotpy-wpilib #605: Explore pybind11 around wpilibc

virtuald, 2019-05-19:
RobotPy is a lot of effort to maintain. As more and more third party vendors appear and as the libraries gain more functionality, it's difficult to continue translating Java into Python year after year, and there are often difficult to address performance issues that arise from doing so.

We've had a lot of success with autogenerating pybind11-based wrappers for C++ code for various things, and I think it's worth investigating if we should move to a mostly-wrapper based approach (of course, there are things that would be best done in pure python). If successful, this would potentially make it easier to create wrappers for third party vendor libraries (though we might need to redo our existing wrappers).

While there's definitely an upfront investment required, I feel like it would be less work in the long run. robotpy-cscore is a great example of this -- it doesn't take a lot of effort to update it each year because we don't care about how the internals work, just need to maintain the interface API.

There are definitely packaging issues that would need to be solved, particularly with regards to loading shared libraries on all of our supported platforms. However, if we depend on upstream wpilibc, they're already building those, so we would 'just' need to make the right wheels and it should work.

One really big issue is that moving to wrappers would cause us to lose our simulation capability. This is by far the best feature of RobotPy, and losing it would be a Bad Thing. However, there's a potential here for creating a backend using C++/HAL that could be usable from all languages -- and that would be a net benefit for everyone.

Moving to wrappers would allow us to benefit from any desktop HAL simulation support that third party vendors create in the future -- which if a C++/Java backend existed, they would be more likely to do.

auscompgeek, 2019-05-19:
Note that this would mean that pynetworktables would no longer be pure-Python, as wpilibc depends on the C++ ntcore. This will likely make using NetworkTables with Python on a coprocessor to be slightly more painful than currently (although I plan to maintain my packaging to compensate for this on the more common platforms).

Another thing is that this will mean simulation will only be supported on whatever upstream WPILib supports, whereas the RobotPy simulation works on whatever Python runs on. It'll also be more painful to install simulation stuff on anything that PyPI doesn't support wheels for (i.e. non-x86 platforms and non-glibc *nix). Dunno whether this would be a deal-breaker for anyone - hopefully people will speak up if so.

The other big thing for me is that this approach will mean duck typing will no longer be possible for anything that gets passed into our wpilib package. Probably not a huge deal-breaker, but something we might need to consider.

One thing we will need to double-check is whether subclassing across pybind11 modules works as expected, `isinstance` and all. If not, we will have to use something else, perhaps shiboken2.

(It's also possible that using shiboken2 might be easier than trying to manipulate our existing pybind11 wrapper generation to behave correctly for wpilibc. I'm not confident that the stuff in robotpy-rev will actually work correctly as is against wpilibc.)

PeterJohnson, 2019-05-19:
Coming full circle, considering that the original RobotPy was SIP-generaged wpilibc wrappers: https://github.com/robotpy/robotpy-crio/tree/2014/Packages/wpilib/sip

virtuald, 2019-05-19:
Yep! And if you recall, I was skeptical of pure-python wpilib then too. It was a good experiment, and for the most part it's good enough -- but it's so. much. work. Work that just isn't interesting.

Additionally, pybind11 is much easier to work with IMHO than SIP, and things like the advent of wheels have made some of the disadvantages of wrappers less than they used to be.

ThadHouse, 2019-05-19:
Regarding the native builds, we do have the CMake build setup that can build basically on any platform, and for just ntcore works even without OpenCV with the right flags, and can ignore Java too. By enabling any pybind builds to detect cmake installed libraries (the build can install them too), you should in theory be able to easily support any platform, not just the base ones we support.

Note for 2020 we will likely be adding upstream support for Jetson TX2 and Jetson Nano, and we already have Pi builds, so it would only be weird other platforms that would need the custom builds.

auscompgeek, 2019-05-19:
pybind11 does have a CMake-based build, so if we're going to do CMake that'd be the way to go.

virtuald, 2019-05-19:
Historically, I've wanted 'pip install foo' to just work, so I've avoided using cmake+pybind11. However, that may need to be revisited.

auscompgeek, 2019-05-19:
pybind11 also have an example where they call cmake from setup.py, so that shouldn't be an issue for source builds: https://github.com/pybind/cmake_example/blob/master/setup.py

ThadHouse, 2019-05-19:
Yeah, at least for non supported by WPILib platforms. It would likely be best to use the wpilib builds for the platforms we support, to ensure compatibility with vendors, but for other platforms its not a bad solution.

Can you make setup.py do different things on known vs unknown platforms?

And on unknown platforms, I don't think CMake and GCC is an unreasonable requirement. On Windows CMake and MSVC is unreasonable, but binary blobs are much easier on windows then other platforms, because they do actually just work.

auscompgeek, 2019-05-19:
setup.py is arbitrary code, so we can make it do anything we want to :wink:

auscompgeek, 2019-05-19:
Something else that we will need to consider if we go down this path is generating documentation, and being able to generate typing stubs.

auscompgeek, 2019-06-08:
Another random thought: a bunch of methods take out params, and some also take raw pointers to arrays with a separate size param. We will want to make sure these methods have the same method signature our existing pure-Python code has. Doing this with our pybind11 generation scripts may be nontrivial.

auscompgeek, 2019-07-02:
Thinking aloud: Could https://github.com/google/clif be a better option than pybind11 here? hmm... [doesn't support overloading](https://github.com/google/clif/blob/master/clif/python/faq.md#q-swig-supports-overloaded-c-functions-how-do-i-do-that-in-clif)

virtuald, 2019-09-30:
I'd like to give this a go. I've started by porting ntcore -- it works, but still needs quite a bit of polish. I think there are a few things that are needed:

* I'd like to autogenerate wrappers as much as possible. Currently our wrapper stuff is copy/pasted to mulitple repos, so a new project with a pybind11 autogenerator would be needed. I think most of what is needed is now part of the ntcore binding generator (which was derived from the -rev bindings generator).
  * One thing that would be good is to generate .pyi files, which I've omitted ... but maybe https://github.com/sizmailov/pybind11-stubgen could do the job?
* I think each of the upstream wpilib projects would need it's own python project (wpiutil, etc), and each of them would need to package their headers so that other builds can use them (the same way that pybind11 headers are packaged)
* Builds would still be setup.py/distutils -- it would be great if we could do cross-platform builds though, then we could upload ipk files for roborio via CI
  * Though, we should consider whether there's a better way to do this
* PyPi uploads would consist of a source package, and .whl that embed stuff for our target platforms (py36/37/38 win32/64, linux64 multiarch, osx). Anyone targeting anything else would have to build their own. The wheels would embed upstream wpilib/vendor shared libraries in them as published on maven
  * Ideally, we would get this working once, and the rest would be copy/paste CI configurations

Regarding backwards compatibility -- with a transition such as this, I'm ok with minor breakage, and this change should result in less incompatibilities in the future?

virtuald, 2019-09-30:
An open question I'm undecided on is what to do with pynetworktables. It's useful to have a pure-python NetworkTables implementation for coprocessors, so I don't think I want it to go away. I'm tempted to create a new package called 'pyntcore', and have it export an `ntcore` package which does essentially the same thing as `networktables` package now (except it would be binary wrappers). pynetworktable's ntcore package would be moved to `networktables._impl`, which is probably where it should have been anyways.

virtuald, 2019-09-30:
Also, given https://github.com/wpilibsuite/allwpilib/pull/1910, I expect that by migrating to wrappers we won't lose simulation.

PeterJohnson, 2019-09-30:
Re: pyntcore, could you have pynetworktables use pyntcore for the implementation if it's available (with the Python version as a fallback)?

virtuald, 2019-10-10:
I've completed an initial prototype for ntcore, would welcome feedback from anyone who is interested. I expect this will be pretty easy to extend to other RobotPy projects now that I've got the initial hard work done.

https://github.com/robotpy/robotpy-build is the initial build system

And the two libraries that are needed to compile binary ntcore bindings:

* wpiutil: https://github.com/robotpy/robotpy-wpiutil
* pyntcore: https://github.com/robotpy/pyntcore

virtuald, 2019-10-24:
An initial autogenerated hal implementation has been finished: https://github.com/robotpy/robotpy-experimental-wpilib

virtuald, 2019-10-25:
I started working on wpilibc tonight, and I've hit the first  problematic roadblock so far. wpilibc depends on the cscore library, which depends on opencv. If we convert our cscore library to robotpy-build (we should), we still have the opencv problem -- which I refer you to https://github.com/robotpy/robotpy-cscore/issues/4 and https://github.com/skvark/opencv-python/issues/22 as open problems regarding bundling OpenCV shared libraries in a python package.

I suppose we could bundle the wpilib opencv as a separate python package, but we would need to build the python module for each platform, could get messy.

Another thought is that anaconda has support for binary and python packages, we could move to using anaconda... but my impression of conda is that the learning curve is a bit higher than just stock python, and I'm not sure it would necessarily avoid this problem anyways.

auscompgeek, 2019-10-25:
Could we just avoid all the wpilibc parts that use cscore altogether? It's just CameraServer, right? We could probably static link wpilibc and hopefully that'd avoid any linkage issue.

virtuald, 2019-10-25:
Maybe. We would still need to have a libwpilibc.so for ctre/etc to link to.. so we could construct a new shared library from the static library? That sounds... terrible.

ThadHouse, 2019-10-25:
Yeah, I did that for c#. It wasn't pretty.

It wouldn't actually be that difficult to remove the opencv dependency from wpilibc. We could dynamically load the symbol, as iirc we only use a single function call that loads cameraserver, that then loads cscore.

auscompgeek, 2019-10-25:
From memory, doesn't RobotBase also register some shared functions with CameraServer to determine its environment or something? How can we reconcile this?

ThadHouse, 2019-10-25:
That's the function I'm talking about. It's a single function call to do all of that, so we can likely dlopen the library and get an fp of that. And just do nothing on missing library, as it's not required to run.

auscompgeek, 2019-10-25:
Ah, that's probably reasonable then.

virtuald, 2019-10-25:
My goal is to first implement 2019.4.1 wrappers, and then upgrade to beta, to prove to myself how much work a year-to-year upgrade could be. 

@ThadHouse if you can make cscore/cameraserver a dynamic dependency in beta, that would be fantastic. I can just hack around the cscore stuff for 2019.4.1 if it's just a temporary thing.

virtuald, 2019-10-25:
Thad has provided a PR for beta: https://github.com/wpilibsuite/allwpilib/pull/1989

auscompgeek, 2019-11-26:
@virtuald How is this going? Last I recall you mentioned something about CppHeaderParser giving you a bit of grief with WPILibC's headers.

virtuald, 2019-11-27:
Well. Me and CppHeaderParser are not doing well. We're fighting.

As reported on gitter, I experimented with [autowig](https://github.com/StatisKit/AutoWIG). It's really cool, but it's slow, has enormous dependencies, and there are aspects of it's wrapper generation that I'm not happy with -- but it does most of what we want.

I'd really like a modified version of AutoWIG that could use CppHeaderParser's output -- AutoWIG's C++ AST seems like it's really well thought out. Additionally, AutoWIG is pluggable, so this is totally doable. This last weekend while traveling I started a notional rewrite of CppHeaderParser because I'm tired of dealing with it's crap and I wanted to see how far I could take it. It's... doable, though maybe not with the time I have to dedicate to the problem.

I need to play with the wrappers that I already have (like pyntcore) and see how they fail/perform, but I haven't been able to find time to do that.

virtuald, 2019-12-19:
So it's 17 days before kickoff, here's my current status:

Essentially, I've wasted a month and haven't really reached out to anyone for help.

I tried writing a new C++ parser, and got bogged down in the details because I was trying to make it perfect -- while I made a lot of progress and it was quite interesting to work on, there's no way this will be done in time, despite my optimism. I did rewrite some pieces that could be imported to CppHeaderParser -- in particular I've got improvements to the lexer which should make comment extraction better, some pieces of my new parser that could be applicable to CppHeaderParser, and a lot of lessons about the ambiguity of C++. I still really hate working with CppHeaderParser, but it is functional for many things even if it can't parse everything in wpilib.

More thoughts:

* AutoWIG is slow and I don't like some of it's choices. However, it does seem to work as long as you're using Conda. I think CppHeaderParser could be adapted to use it's AST, which could be less work than writing a new C++ parser.
  * This would require a bit of work.
* SIP 5.0 is out, and they seem to have a modern build system, and we've used it before successfully
  * Would need to manually create SIP files and figure out how to build them
* It's not too late to make the pure python upgrade happen, particularly with community involvement
* A hybrid approach is possible: move HAL and/or ntcore to pybind11 to get the performance we need, do the rest in pure python
  * It's not clear the wpilib kinematics for 2020 is doable without numpy, but we could package them separately. Additionally, we could extract the C++ version into it's own library and wrap them using pybind11.

It's not too late yet, but it will be soon, and for that I apologize. I need help to decide on the right direction, as I don't really like any of these choices.

auscompgeek, 2019-12-19:
> It's not clear the wpilib kinematics for 2020 is doable without numpy

[I started porting the kinematics stuff to pure-Python](/auscompgeek/wpilib.kinematics). At least swerve does indeed require numpy if we want any semblance of not-abominably-unmaintainable code, but we can probably just keep it in a separate module that isn't imported by default.

> Additionally, we could extract the C++ version into it's own library and wrap them using pybind11.

We'd have to work out how we want to wrap units to actually do this.

virtuald, 2019-12-19:
Other alternatives suggested:

* (from @ArchdukeTim) use robotpy-build to generate wrappers for things that can be generated, but write handwritten wrappers for the rest instead of trying to automate everything
* Autogenerate SIP files via robotpy-build and wrap their PEP 517 build backend
  * This would be a lot easier than writing all of our pybind11 wrappers manually, as it already has a lot of the behaviors we need -- would need to investigate units wrapping and other tricky pieces

virtuald, 2020-01-19:
This has been done.

closed 2020-01-19 by virtuald