robotpy/robotpy-wpilib #606: Drop Python 3.5 support?

auscompgeek, 2019-07-01:
Python 3.6 has variable annotations, which would be super-helpful for type hinting data classes.

PyPy3 has 3.6 support now (with a beta label - I hear coverage.py is currently a bit broken), so not too much of a concern there in case anyone (we?) wanted to go down that road.

In terms of distro support:
- Debian 9 (Stretch) is 3.5, but 10 (Buster) will be released in the next week with 3.7.
- The only supported Ubuntu release still on 3.5 is 16.04 (Xenial).
- RHEL7 has Python 3.6, of all distros...
- All supported Fedora releases are on 3.7.
- openSUSE Leap 42.3 has 3.4, which we don't support anyway. Leap 15.0-15.1 is 3.6.

virtuald, 2019-07-14:
I'm supportive of this.

auscompgeek, 2019-07-15:
I guess someone should make the PRs to remove 3.5 from CI. I'd like to see the magicbot examples updated to use the variable annotation syntax.

M-Shadow, 2019-10-18:
Think that should be about it. Any other repos that need updating to CI/updating in general based on this?

virtuald, 2019-10-18:
pyfrc, pynetworktables, navx, rev... we have a lot of repos. 

FWIW, robotpy-build uses f-strings, so that's definitely going to be 3.6 only.

auscompgeek, 2019-10-18:
We should also make sure the setup.py files list the 3.8 Trove classifier and not the 3.5 one as well.

virtuald, 2020-01-19:
Done.

closed 2020-01-19 by virtuald