robotpy/robotpy-wpilib #616: Add cscore/cameraserver integration back in

virtuald, 2020-01-19:
Should just be copy/paste from the old repo. Need to build/install cscore for roborio too.

ArchdukeTim, 2020-01-19:
Can this not be autogenerated?

auscompgeek, 2020-01-20:
No, we want to keep our cscore out of main process. Also we don't want cscore to be a dependency here otherwise packaging will be impossible.

amoose136, 2020-01-21:
Without  some way to get CSCore I can't run the example multiCameraServer.py that comes with wpilibsuite/FRCVision-pi-gen locally on a windows installation.

virtuald, 2020-01-21:
@amoose136 running cscore locally is actually an annoying proposition, see https://github.com/robotpy/robotpy-cscore/issues/17 ... though, if we packaged OpenCV with robotpy-build, I wonder if we could pull it off?

amoose136, 2020-01-23:
I may have to try the conda thing. I don't really like conda as I don't think it solves the problems it's designed to solved so I'd rather just use stock python.  I was able to get opencv to compile using visual studio community edition but that didn't fix the missing header error when building CSCore and the wheel file linked doesn't actually work anymore either because the version of opencv I built was 4.x series not 3.x series.

commit 0a586f10 (2020-02-04): Add cameraserver shim back in

- Fixes #616
closed 2020-02-04 by virtuald