robotpy/robotpy-wpilib #617: Code will not run, only throwing Requirement.parse errors

JDuskey, 2020-01-23:
The driver station will not show any attempt to try to run code, but when I deploy with the --nc option, I get a bunch of non-stop errors like `Requirement.parse('robotpy-hal<2021.0.0, >=2020.1.2.2')`

auscompgeek, 2020-01-23:
It looks like you've managed to break your RobotPy installation in some way. I would reinstall RobotPy on your robot.

JDuskey, 2020-01-23:
That was the first thing I tried, same result with a fresh image of robotpy.

JDuskey, 2020-01-23:
Also the current version of hal wouldn’t meet this requirements, as the feed has 1.2.1 listed as the latest, but this is asking for at least 1.2.2

auscompgeek, 2020-01-23:
Ah, I see what's happened now. If you install python38-wpilib 2020.1.2.2 instead it should work. It looks like there weren't any real changes in our wpilib 2020.1.2.3 release.

@virtuald could you publish python38-robotpy-hal 2020.1.2.2 and python38-robotpy-wpiutil 2020.1.2.2 on the opkg repo to fix this?

virtuald, 2020-01-23:
Sorry about this! Working on automating our build process so this doesn't happen in the future. Running a build now, will upload it in a few minutes.

virtuald, 2020-01-23:
Should be fixed, thanks.

closed 2020-01-23 by virtuald