robotpy/robotpy-wpilib #626: Add feet/degrees to various geometry/trajectory things

virtuald, 2020-02-05:
Internally everything uses meters/radians. In C++ this isn't a big deal because the conversions are done for you automatically. For us, we have to pick statically. My inclination is to add static methods:

```
Pose2d.fromFeet(12)
```
Etc. Not sure about the right naming for this.

MrRSquared, 2020-02-10:
Hello, I see this is closed.  Did you all add feet/second?
If not, I would like this, but can we not just translate them in code ourselves? 
input = x*3.28084 #x is the feet we want to input into the function.
output = y/3.28084 #y is the feet we want to get out of the function.

auscompgeek, 2020-02-10:
Where do you see this is closed? I still see the issue open.

I'm personally inclined to do what WPILib did for Java: keep everything in metres, and provide convenience conversion functions between units (which I think we might already do in robotpy-wpilib-utilities).

virtuald, 2020-02-10:
I would be fine with you adding a scalar version that has the original name. I don't really want to remove them at this time, because we've already broken users enough times this season.

MrRSquared, 2020-02-10:
I think that sounds like the best way from my limited experience point of view.
As for closed, I swear when I looked earlier, the two buttons on the top of the page were the red Closed buttons.  Now, it is just the linked mention in #618.  So, definitely still open.

virtuald, 2020-03-24:
The way I've decided to deal with this is by adding various getters/setters and static methods that take non-SI units where it makes sense. This has seemed like a sensible compromise to me, and I'm pretty happy with it so far.

closed 2020-03-24 by virtuald