robotpy/robotpy-wpilib #631: Tracebacks double up

auscompgeek, 2020-02-06:
When a Python exception occurs, the traceback that gets printed repeats itself in its entirety. Example:

```pytb
Traceback (most recent call last):
  File "~/.local/share/virtualenvs/rpy-dev/lib/python3.8/site-packages/wpilib/_impl/start.py", line 106, in start
    self.robot.startCompetition()
  File "robot.py", line 16, in robotInit
    wpilib.SmartDashboard.putData("test", self.test)
  File "~/.local/share/virtualenvs/rpy-dev/lib/python3.8/site-packages/wpilib/_impl/start.py", line 106, in start
    self.robot.startCompetition()
  File "robot.py", line 16, in robotInit
    wpilib.SmartDashboard.putData("test", self.test)
RuntimeError: return_value_policy = copy, but type frc::SendableBuilderImpl is non-copyable!
```

virtuald, 2020-02-07:
I think this is because DriverStation.report is printing it, and so is HAL.

virtuald, 2020-02-07:
https://github.com/robotpy/robotpy-wpilib/blob/615892636695b254d02bf7021139f3f1fe8a773f/wpilib/_impl/report_error.py#L54

auscompgeek, 2021-01-19:
That argument is `printMsg`, right? The traceback in the OP is in simulation.

virtuald, 2022-04-20:
Fixed in a6b96b0274cef7fd8c91eae41afd77a1d84ea066?

closed 2022-04-20 by virtuald