robotpy/robotpy-wpilib #633: Double-allocating resources does not raise an exception

auscompgeek, 2020-02-07:
It did in previous seasons. This makes it a bit too easy to not notice until you check the console, since the messages from WPILibC seem to not show when the DS mini-console is set to error level.

virtuald, 2020-02-23:
I tried customizing the trampolines, but realized after I was done that the trampolines are only used when python classes inherit the object. :(

virtuald, 2020-02-23:
I think this is something we won't be able to successfully do until 2021, because it would require hooking errorbase in the underlying library.

virtuald, 2020-02-23:
An alternative for teams that care (maybe we should do this in Magicbot?) is do the following at the end of robotInit:

```
errors = wpilib.ErrorBase.getGlobalErrors()
if errors:
   # print them...
   raise ValueError("OH NOSE")
```

virtuald, 2022-01-15:
This is fixed in 2022.

closed 2022-01-15 by virtuald