robotpy/robotpy-wpilib #637: Unable to import trajectory or controller

adam-mcdaniel, 2020-02-08:
[image]

adam-mcdaniel, 2020-02-08:
Our team needs to be able to access the RamseteController object. In the previous RobotPY version, it was in the main package scope, but now Im under the impression that its in the controller module

virtuald, 2020-02-08:
Oh, I see what happened. Sorry about that, @auscompgeek renamed them. Will push an updated release in a few minutes.

virtuald, 2020-02-08:
Fixed in 85f1f6f1e999d4055bda6a7198f8cdf5d6a6c557

closed 2020-02-08 by virtuald

virtuald, 2020-02-08:
Once 2020.2.2.7 finishes building (maybe 10 minutes?), this fix should be available for simulation. Just uploaded for roborio.