robotpy/robotpy-wpilib #640: TrajectoryState class has no attribute "curvature"

BrandonWerbel, 2020-02-08:
The [TrajectoryState](https://github.com/robotpy/robotpy-wpilib/blob/85f1f6f1e999d4055bda6a7198f8cdf5d6a6c557/gen/trajectory/Trajectory.yml#L24) class as it's generated has no curvature attribute. This is necessary when we sample the trajectory and use the curvature value for calculations. Here's the curvature value in the C++ libraries as a reference: [Trajectory.h](https://github.com/wpilibsuite/allwpilib/blob/1b85066d267d46ad0d7c699140e2ba335a90cf9a/wpilibc/src/main/native/include/frc/trajectory/Trajectory.h#L53)

virtuald, 2020-02-08:
Oops. Are there any other attributes missing that you need?

closed 2020-02-08 by virtuald

BrandonWerbel, 2020-02-08:
Could you please add a constructor for ChassisSpeeds that takes in the arguments: (vx, vy, omega). Right now the constructor is blank and the attributes are not settable.

virtuald, 2020-02-08:
The attributes should be settable in 2020.2.2.8.

commit 13a4e1b9 (2020-02-08): Add curvature attribute

- Fixes #640