robotpy/robotpy-wpilib #649: Reported segfault in SmartDashboard.putStringArray

virtuald, 2020-02-14:
/cc @celttechie 

Windows 10, Python 3.8.1 64-bit. Might be https://github.com/robotpy/robotpy-wpiutil/issues/7?

auscompgeek, 2020-02-14:
Quite possibly if the strings are computed. I imagine constants would work by chance.

auscompgeek, 2020-02-14:
Actually, hmm. ntcore stores a `std::vector<std::string>` and references that. vector and string own their contents, right?

virtuald, 2021-02-16:
I looked at this a bit deeper, and I haven't been able to reproduce. I think if this were to happen, it would be that somehow an invalid py::str was returned from a function somewhere and added to an array that was used. Lots of pybind11 churn this summer, including a bunch of string stuff, so it's likely this has been addressed.

* https://github.com/pybind/pybind11/pull/2477
* https://github.com/pybind/pybind11/pull/2349

closed 2021-02-16 by virtuald