robotpy/robotpy-wpilib #666: Design physics/simulation support

virtuald, 2020-02-20:
Now that we have https://github.com/wpilibsuite/allwpilib/pull/2377 ... this provides a convenient mechanism to hook in our physics.py support. I like having it as a separate file that automatically gets loaded in simulation.

I'm thinking that we hide those two methods from users, and call something internal instead... but then I was thinking maybe I try to detect if the user overrode them, and if not go back to our old physics.py stuff? Or we just make users use simulationInit/simulationPeriodic manually.

One flaw that I hadn't thought through is that simulationPeriodic doesn't provide the elapsed time, which is needed for physics computations.

auscompgeek, 2020-02-20:
> One flaw that I hadn't thought through is that simulationPeriodic doesn't provide the elapsed time, which is needed for physics computations.

That's not too bad. We can call hal.getFPGATime.

virtuald, 2020-03-02:
Implemented in https://github.com/robotpy/pyfrc/pull/170

closed 2020-03-02 by virtuald