robotpy/robotpy-wpilib #667: Add Swerve Drive support

virtuald, 2020-02-23:
We have templates, should be doable now. I think we probably should just suffix with the number of modules supported, and support common configurations.

My understanding is 3, 4, and 6 modules are probably most likely.

closed 2020-02-23 by virtuald

KenwoodFox, 2022-01-10:
Was this completed?

virtuald, 2022-01-10:
Yes, that's why the ticket says "closed in 4df5129"

KenwoodFox, 2022-01-10:
Awesome! Thanks so much! Should have looked more closely, it looked like automation closed the issue but now i see that you did.

virtuald, 2022-01-10:
There's a bunch of docs at https://robotpy.readthedocs.io/projects/wpimath/en/stable/wpimath.kinematics.html

KenwoodFox, 2022-01-10:
> There's a bunch of docs at [robotpy.readthedocs.io/projects/wpimath/en/stable/wpimath.kinematics.html](https://robotpy.readthedocs.io/projects/wpimath/en/stable/wpimath.kinematics.html)

This is great, thank you so much!

commit 4df5129d (2020-02-23): Add support for swerve drive kinematics

- Fixes #667