robotpy/robotpy-wpilib #674: Auto-generated type annotations?

bvisness, 2020-07-09:
Our team is exploring switching from Java to Python sometime in the coming seasons, and we're exploring how type checking can fit into that. We've had some success getting mypy working with handwritten type stubs, but that gets old fast.

I noticed that the robotpy docs have robust types for everything, including third-party packages. What would it take to generate type stubs (or type-annotated Python code) with that same information?

I was digging into this myself a bit early this year, but got a bit lost trying to find out how the docs are built. I'm absolutely willing to work on this myself if somebody can point me in the right direction.

virtuald, 2020-07-09:
The right way to do this is to modify/leverage robotpy-build to generate the type stubs. Some investigation has already been done on this, but definitely would be interested in someone finishing up the work. See https://github.com/robotpy/robotpy-build/issues/1 .

closed 2020-07-09 by virtuald

virtuald, 2020-07-10:
If you want to tackle this problem, I'd be happy to chat with you (either on our Gitter channel or via some voice like Google Meet/whatever) sometime about how the docs are built and some of the challenges that we've faced regarding type annotations. Let me know -- I suspect this would be a much appreciated feature by RobotPy teams!