robotpy/robotpy-wpilib #681: 2021 - import wpilib and sim issues after  pip install -U robotpy[all]

aesatchien, 2021-01-22:
Hello, 
We've been using the sim on 2020 with no problem for at least a year now.
I tried the 2021 release yesterday on a fresh conda environment with Python 3.8.6, doing the `pip install robotpy` and then double checking with robotpy[all] (and ipython for some troubleshooting).

Running the simple examples/physics/ with the standard `python robot sim` works fine in 2020 but gives me the following error in 2021: (AttributeError: type object 'wpilib._wpilib.SmartDashboard' has no attribute 'Field2d') which looks like the right place since they moved the Field2d() to the SmartDashboard for 2021.  Full error message and output of pip list attached.

A quick check of importing wpilib - autocomplete thinks wpilib.SmartDashboard is there but trying to import it  ModuleNotFoundError: No module named 'wpilib.SmartDashboard'.

Thanks for all of the great work you do! -Cory

[field2d_error.txt](https://github.com/robotpy/robotpy-wpilib/files/5857226/field2d_error.txt)
[piplist.txt](https://github.com/robotpy/robotpy-wpilib/files/5857254/piplist.txt)

virtuald, 2021-01-22:
Sorry about that, been taking a day or two break after getting everything out for 2021, haven't had time to fix the documentation/examples/etc yet.

Fixed in https://github.com/robotpy/pyfrc/commit/aa55cbfdce410d7d8a48f0f09d47cb04f9493911 and in the example at https://github.com/robotpy/examples/commit/044779929039cf74724e9d7932e9b0e465fad9c7 ... a new build should be on pypi within a few minutes.

Would love some feedback on how difficult the 2020 -> 2021 transition goes. The primary issue with simulation is that all of the hal.simulation stuff is now in wpilib.simulation, and the names are slightly different. See the docs at https://robotpy.readthedocs.io/projects/wpilib/en/stable/wpilib.simulation.html

closed 2021-01-22 by virtuald

aesatchien, 2021-01-22:
Thanks!  I have the new build running the 2021 sim on that sample bot.  The new sim widgets are great.   I can't figure out how to set the field image, and some of the widgets don't seem to come up yet, but I'm sure I'll get there.  Thanks for the link to all of those new names.  Once we transition more of our 2020 sim code to 2021 I'll send some more feedback.