robotpy/robotpy-wpilib #691: wpilib.shuffleboard not implemented

virtuald, 2021-02-22:
There's a virtual base class in there somewhere which made this tricky to do correctly, though I don't quite remember why that is.

virtuald, 2022-10-03:
I have something for this, but it needs testing.

auscompgeek, 2022-12-27:
We currently have a TODO comment:

https://github.com/robotpy/robotpy-wpilib/blob/69cf7d37c1bff7d278464a091ced1a6154ec0ba0/pyproject.toml#L282

Reopening to track this.

auscompgeek, 2022-12-27:
@virtuald do you recall how exactly it's broken? I presume it's not related to https://github.com/wpilibsuite/BetaTest/issues/80

virtuald, 2022-12-27:
I feel it was some NT4 thing, but I imagine that's been fixed by now. We should re-enable and add a simple test to make sure it works.

commit ff5447ea (2022-10-03): Add shuffleboard wrappers

- Fixes #691
- Fixes #705
.. not sure if they work
commit 65d0c4fc (2022-10-23): Add shuffleboard wrappers

- Fixes #691
- Fixes #705
.. not sure if they work
closed 2023-11-13 by virtuald