robotpy/robotpy-wpilib #692: `import wpilib` in repl kills interpreter

TheTripleV, 2021-05-21:
Importing wpilib in a repl on the roborio causes the python interpreter to be killed.

auscompgeek, 2021-05-22:
Anything interesting in `dmesg`?

TheTripleV, 2021-05-24:
Nope. It only happens as lvuser. Doesn't happen as admin though.

In both cases, it does kill and restart the robot code under a new pid.

The behavior I expected was that the robot code would be killed and the repl would stay.

auscompgeek, 2021-05-24:
I think this is expected behaviour when you SSH in as lvuser. From what I've heard, the process that runs automatically as frcUserProgram has more privileges than a regular lvuser process (despite it also running as lvuser). Someone from the WPILib team could probably confirm.