robotpy/robotpy-wpilib #693: `OSError: Failure` on Linux

Mockapapella, 2021-09-13:
I managed to program my RoboRIO to respond to a throttle on Windows using the USB cable connection. I'm now trying to do the same on Linux, however I'm deploying the code to the robot using the ethernet connection due to issues I was having with the USB connection. I'm getting an `OSError: Failure` when the paramiko sftp session closes. Here is my `robot.py` code:

```
#!/usr/bin/env python3

import wpilib
import ctre

class MyRobot(wpilib.TimedRobot):

    # update every 0.005 seconds/5 milliseconds (200Hz)
    kUpdatePeriod = 0.005

    def testInit(self):
        """Robot initialization function"""

        self.motor = ctre.WPI_TalonSRX(0)  # initialize the motor as a Talon on channel 0
        self.joyStick = wpilib.Joystick(0)  # initialize the joystick on port 0

    def testPeriodic(self):
        """Runs the motor from a joystick."""

        # Set the motor's output.
        self.motor.set(((self.joyStick.getThrottle() + 1) / 2) * 0.25)

if __name__ == "__main__":
    wpilib.run(MyRobot)
```

Everything is installed in a virtual environment (`venv`) initialized with python3. When I run `python robot.py deploy` it finds the robot on my network but then returns an `OSError` at the same point as when it normally completes successfully on Windows:

```
10:17:59:901 INFO    : wpilib              : WPILib version 2021.3.1.0
10:17:59:901 INFO    : wpilib              : HAL version 2021.3.1.0
10:17:59:901 INFO    : wpilib              : Running with simulated HAL.
10:17:59:901 INFO    : wpilib              : robotpy-wpiutil version 2021.3.1.0
10:17:59:901 INFO    : wpilib              : robotpy-wpimath version 2021.3.1.0
10:17:59:902 INFO    : wpilib              : robotpy-halsim-gui version 2021.3.1.0
10:17:59:902 INFO    : wpilib              : robotpy-ctre version 2021.2.2
10:17:59:902 INFO    : wpilib              : pyntcore version 2021.3.1.0
10:17:59:902 INFO    : faulthandler        : registered SIGUSR2 for PID 66025
ERROR: large files found (larger than 250000 bytes)
...
Upload anyways? [y/n]y
10:02:40:061 INFO    : robotpy.installer   : Finding robot for team 1716
10:02:40:066 INFO    : robotpy.installer   : -> Robot is at 192.168.1.81
10:02:40:066 INFO    : robotpy.installer   : Connecting to robot via SSH at 192.168.1.81
10:02:40:229 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_7.6)
10:02:40:436 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
10:02:40:436 INFO    : paramiko.transport  : Authentication (password) successful!
RoboRIO has WPILib version 2021.3.1.0
10:02:44:382 INFO    : paramiko.transport.sftp: [chan 5] Opened sftp connection (server version 3)
make /home/lvuser/py_new
requirements.txt -> /home/lvuser/py_new/requirements.txt
client.py -> /home/lvuser/py_new/client.py
robot.py -> /home/lvuser/py_new/robot.py
make /home/lvuser/py_new/venv
venv/pyvenv.cfg -> /home/lvuser/py_new/venv/pyvenv.cfg
make /home/lvuser/py_new/venv/include
make /home/lvuser/py_new/venv/share
make /home/lvuser/py_new/venv/share/python-wheels
venv/share/python-wheels/progress-1.5-py2.py3-none-any.whl -> /home/lvuser/py_new/venv/share/python-wheels/progress-1.5-py2.py3-none-any.whl
...
venv/lib/python3.8/site-packages/wpimath/_impl/lib/libwpimath.so -> /home/lvuser/py_new/venv/lib/python3.8/site-packages/wpimath/_impl/lib/libwpimath.so
10:04:11:174 INFO    : paramiko.transport.sftp: [chan 5] sftp session closed.
Traceback (most recent call last):
  File "robot.py", line 40, in <module>
    wpilib.run(MyRobot)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/wpilib/_impl/main.py", line 193, in run
    retval = options.cmdobj.run(options, robot_class, **kwargs)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/pyfrc/mains/cli_deploy.py", line 198, in run
    if not self._do_deploy(ssh, options, robot_filename, robot_path):
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/pyfrc/mains/cli_deploy.py", line 372, in _do_deploy
    ssh.sftp(py_tmp_dir, deploy_dir, mkdir=not options.in_place)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/robotpy_installer/sshcontroller.py", line 120, in sftp
    sftp.put(str(local_fname), str(remote_fname))
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_client.py", line 759, in put
    return self.putfo(fl, remotepath, file_size, callback, confirm)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_client.py", line 716, in putfo
    size = self._transfer_with_callback(
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_client.py", line 679, in _transfer_with_callback
    writer.write(data)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/file.py", line 405, in write
    self._write_all(data)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/file.py", line 522, in _write_all
    count = self._write(data)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_file.py", line 208, in _write
    t, msg = self.sftp._read_response(req)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_client.py", line 865, in _read_response
    self._convert_status(msg)
  File "/home/quinten/Documents/Software_Development/Robotics/Robot-Code-Testing/venv/lib/python3.8/site-packages/paramiko/sftp_client.py", line 898, in _convert_status
    raise IOError(text)
OSError: Failure
```

requirements.txt:
```
astroid==2.7.3
atomicwrites==1.4.0
attrs==21.2.0
backports.entry-points-selectable==1.1.0
bcrypt==3.2.0
black==21.8b0
cffi==1.14.6
cfgv==3.3.1
click==8.0.1
colorama==0.4.4
coverage==5.5
cryptography==3.4.8
distlib==0.3.2
filelock==3.0.12
identify==2.2.14
iniconfig==1.1.1
isort==5.9.3
lazy-object-proxy==1.6.0
mccabe==0.6.1
mypy-extensions==0.4.3
nodeenv==1.6.0
packaging==21.0
paramiko==2.7.2
pathspec==0.9.0
Pint==0.17
platformdirs==2.3.0
pluggy==1.0.0
pre-commit==2.15.0
py==1.10.0
pycparser==2.20
pyfrc==2021.1.1
pylint==2.10.2
PyNaCl==1.4.0
pynetconsole==2.0.2
pynetworktables==2021.0.0
pyntcore==2021.3.1.0
pyparsing==2.4.7
pytest==6.2.5
PyYAML==5.4.1
regex==2021.8.28
robotpy==2021.2.8
robotpy-ctre==2021.2.2
robotpy-hal==2021.3.1.0
robotpy-halsim-gui==2021.3.1.0
robotpy-installer==2021.0.3
robotpy-wpilib-utilities==2021.0.1
robotpy-wpimath==2021.3.1.0
robotpy-wpiutil==2021.3.1.0
six==1.16.0
toml==0.10.2
tomli==1.2.1
typing-extensions==3.10.0.2
virtualenv==20.7.2
wpilib==2021.3.1.0
wrapt==1.12.1
```

I'm running this one Ubuntu 20.04

TheTripleV, 2021-09-13:
Hmm. It's trying to copy everything in your venv over. I believe this was fixed by robotpy/pyfrc#184.

I've published a new version of pyfrc including that change. Can you update pyfrc (to 2021.1.2) and try again?

Mockapapella, 2021-09-13:
```
14:08:59:204 INFO    : wpilib              : WPILib version 2021.3.1.0
14:08:59:204 INFO    : wpilib              : HAL version 2021.3.1.0
14:08:59:204 INFO    : wpilib              : Running with simulated HAL.
14:08:59:204 INFO    : wpilib              : robotpy-wpiutil version 2021.3.1.0
14:08:59:204 INFO    : wpilib              : robotpy-wpimath version 2021.3.1.0
14:08:59:204 INFO    : wpilib              : robotpy-halsim-gui version 2021.3.1.0
14:08:59:204 INFO    : wpilib              : robotpy-ctre version 2021.2.2
14:08:59:204 INFO    : wpilib              : pyntcore version 2021.3.1.0
14:08:59:204 INFO    : faulthandler        : registered SIGUSR2 for PID 6762
14:08:59:205 INFO    : robotpy.installer   : Finding robot for team 1716
14:08:59:211 INFO    : robotpy.installer   : -> Robot is at 192.168.1.81
14:08:59:211 INFO    : robotpy.installer   : Connecting to robot via SSH at 192.168.1.81
14:08:59:283 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_7.6)
14:08:59:377 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
14:08:59:377 INFO    : paramiko.transport  : Authentication (password) successful!
RoboRIO has WPILib version 2021.3.1.0
14:08:59:964 INFO    : paramiko.transport.sftp: [chan 5] Opened sftp connection (server version 3)
make /home/lvuser/py_new
requirements.txt -> /home/lvuser/py_new/requirements.txt
client.py -> /home/lvuser/py_new/client.py
robot.py -> /home/lvuser/py_new/robot.py
14:08:59:979 INFO    : paramiko.transport.sftp: [chan 5] sftp session closed.

SUCCESS: Deploy was successful!
```

Worked like a charm. Thanks!

closed 2021-09-13 by Mockapapella