robotpy/robotpy-wpilib #695: Fix Mechanism2D

virtuald, 2022-01-15:
It was changed a lot in 2022, and it's going to be tricky to wrap. Will revisit after the first release.

PeterJohnson, 2022-01-15:
You might want to do pure Python for it if it's tricky to wrap, as it's really just convenience wrappers around sticking stuff into NT.

virtuald, 2022-01-19:
Looked at it tonight, I've got a solution for it. Didn't understand the intent of the API originally.

In particular, it would have been really difficult to allow users to create their own custom MechanismObject2d objects without some kind of append function that would accept an already constructed object (or we would need to constrain the arguments passed to custom object constructors). However, I see that it doesn't really make sense to make your own custom objects with the current API, so that is no longer a problem.

commit ab3fc256 (2022-01-19): Add bindings for Mechanism2d et al

- Fixes #695
commit f863c91f (2022-01-19): Add bindings for Mechanism2d et al

- Fixes #695
PR #698 (2022-01-19): Add bindings for Mechanism2d et al
- Fixes #695
closed 2022-01-19 by virtuald