robotpy/robotpy-wpilib #713: [BUG]: Exceptions from startCompetition don't terminate code on roboRIO

auscompgeek, 2022-05-14:
### Problem description

If an exception is raised from `startCompetition`, e.g. within `testPeriodic`, then the exception is raised but doesn't cause the robot code process to terminate, when on the roboRIO.

### Operating System

RoboRIO

### Reproducible example code

```text
import wpilib

class Robot(wpilib.TimedRobot):
  def testPeriodic(self):
    raise RuntimeError
```

virtuald, 2023-01-03:
I was able to duplicate this once, but then it never happened once I did `raise RuntimeError()` instead (and in fact, it kept working even when I changed it back). Something weird here, for sure, but I don't see any reason why this bug would occur.

virtuald, 2023-11-13:
Closing, can reopen if we see it again.

closed 2023-11-13 by virtuald