robotpy/robotpy-wpilib #721: [BUG]: Error 404 Building Robotpy-halsim-gui

Aldhanekaa, 2023-02-17:
### Problem description

So I tried to install Robotpy 2022 on my Jetson nano which runs on Ubuntu 18.04, it managed to install other else than robotpy-halsim-gui. I choose robotpy 2022 because, [robotpy 2023 doesn't work on Ubuntu 18.04](https://www.chiefdelphi.com/t/python-networktables-and-cscore-on-jetson-nano/423286/3). [
[image] . 

### Operating System

Linux

auscompgeek, 2023-02-17:
robotpy-halsim-gui is not, and never was, available for ARM64 platforms.

virtuald, 2023-02-17:
Yeah, this isn't possible to do for 2022 or 2023 because WPILib doesn't publish halsim-gui for ARM.

virtuald, 2023-02-17:
Actually, they are publishing ARM artifacts now for halsim-gui, but only for 2023 (which isn't going to help you): https://frcmaven.wpi.edu/ui/native/release/edu/wpi/first/halsim/halsim_gui/2023.3.2/

closed 2023-02-17 by virtuald