robotpy/robotpy-wpilib #724: Installation problems

otherjon, 2023-11-09:
New-ish mentor trying to migrate new-ish team to Python, so it's almost certainly my mistake, but hopefully you can point me in the right direction (and maybe there's some documentation to update).

From a brand-new Windows 11 VM, I install WPILib, Python (current 3.12 fails to install robotpy (errors with Path objects), so I'm using 3.11.6 now), Phoenix Tuner, and REV Hardware Client, then I upgrade pip and then run `python3 -m pip install robotpy`.  (I get other problems trying the 2024.0.0b3 version, but that's another story.)  Downloading and installing Python on the roboRIO works fine.  But after I successfully download files with `python3 -m robotpy_installer download robotpy`, attempting to install them on the roboRIO yields a stack trace:
```
 [...]
16:07:40:691 INFO    : robotpy.installer   : -> RoboRIO 2 image version: 2023_v3.1
16:07:40:777 INFO    : robotpy.installer   : -> RoboRIO disk usage 499.6M/29.0G (2% full)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\robotpy_installer\__main__.py", line 4, in <module>
    main()
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\decorators.py", line 45, in new_func
    return f(get_current_context().obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\robotpy_installer\installer.py", line 748, in pip_install
    ssh.exec_cmd(" ".join(pip_args), check=True, print_output=True)
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\site-packages\robotpy_installer\sshcontroller.py", line 94, in exec_cmd
    print(line, end="")
  File "C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'charmap' codec can't encode characters in position 5-44: character maps to <undefined>
```

What am I doing wrong?

TheTripleV, 2023-11-09:
> cp1252

Ah yes. Windows is choosing the wrong encoding probably. More Io should probably explicitly use utf8.

Note: to download beta releases you should use the --pre flag in pip to allow it to download preleases.

otherjon, 2023-11-09:
Am I better off trying to make the existing robotpy module work, or trying to make the prerelease module work?

If the existing one: How do I work around this problem to install successfully?  Do I need to know what "More Io" is?  I can't actually be the first person to try installing robotpy on Windows, am I?  (Well, a Windows VM, anyway -- my host machine is Linux.)

If the prerelease module: Do I need to worry about the warnings that I need the 2024 roboRIO image?  If so, where do I get it?  Should the Chief Delphi post at https://www.chiefdelphi.com/t/robotpy-2024-beta-3-now-available/443705 get an edit, or at least a note about using `--pre` in the `pip install` command?

TheTripleV, 2023-11-09:
No, `pip install robotpy==2024.0.0b3` should work. The alternative is `pip install --pre robotpy` if you weren't specifying the version number.

What error do you get trying to install on 3.12?

And for everything:
1. What terminal are you using? (command prompt / windows terminal)
2. What shell are you using? (command prompt / powershell / bash)
3. What language is your windows configured to?